msinger/dmg-schematics #18: CLOCK: ASOL has swapped inputs and swapped outputs

msinger, 2022-03-14:
For NOR-latch ASOL on clock sheet, compared to furrtek/DMG-CPU-Inside#29 and [here](http://iceboy.a-singer.de/doc/dmg_cells.html#srlatch), the SET&RESET inputs are swapped and also the Q and ~Q outputs are swapped. Since both, inputs and outputs are swapped, this is logically still correct, though. But I think we should keep it consistent. Was this an accident or was there a reason. I can change the pinning in the documentation when necessary.

rgalland, 2022-03-16:
[image]

Isn't it what you see?

msinger, 2022-03-16:
No, I see this:
[image]
The ASOL label already correctly indicates how the cell is rotated. All cells in this row are rotated 90° counter-clockwise compared to the usual vertical columns of cells. It also must be flipped, because the supply rails are flipped for this row.

I understand you did this:
[image]
But then the knobs in the middle don't match, so it can't be rotated clockwise.

rgalland, 2022-03-16:
Incredible. Yes, I think you are right. I'll change is now.

closed 2022-03-16 by rgalland