msinger/dmg-schematics #50: APU_CLK: Label FROM_CPU is wrong

msinger, 2022-03-27:
On APU clock and reset sheet, the label FROM_CPU that goes into EFOP should be T1~{T2}. It is also wrong in Furrtek's schematic. FROM_CPU doesn't exist at all. We also already fixed it for APYS. These were the two occurrences of that label.

rgalland, 2022-03-28:
~{T1}T2 instead?

msinger, 2022-03-28:
Hm, I think T1~{T2} that I wrote down in my notes is the correct one after inverting all T1 and T2 signals. You know, the thing with the inverting input pads...
But yes, since we didn't fix those signals yet, ~{T1}T2 would be correct for consistency. But then APYS also needs to be changed to ~{T1}T2.

rgalland, 2022-03-28:
OK, we will correct this eventually. ~{T1}T2 is still showing on page 25. I'll leave the issue for now until I start making the changes and add a note to modify sheet 25.

closed 2022-05-05 by rgalland