msinger/dmg-schematics #76: PPU: PPU_1MHZ & HCLK & YCLK mixup

msinger, 2022-04-09:
On FF41 STAT sheet, you have a signal called HCLK, which goes into ROPO. HCLK is not generated anywhere (yet?). ROPO should get it's clock from TALU, which you have temporarily placed on the LCD sheet (according to the comment). You labeled the signal that is coming out of TALU as PPU_1MHZ, not HCLK.

Then on LCD sheet, you routed the signal that you called PPU_1MHZ into SYGU and RUTU. But SYGU and RUTU should actually get SONO as clock, which is the inverse of TALU.

Back on FF41 STAT sheet, you routed the *elusive* signal HCLK into the inverter PURE. PURE should actually get the signal you called YCLK, which comes from RUTU on LCD sheet.

I'm totally confused, please send help! :)
I don't know which name you wanted to give to which signal, but clearly it is all scrambled up somehow.

rgalland, 2022-04-10:
I think I have fixed it now. I'm obviously making mistakes while trying to re-organise the sheets and giving sensible names to the nets.

rgalland, 2022-04-10:
I'll commit the changes now. Can you please cast an eye on it before I close the ticket?

msinger, 2022-04-10:
Looks good to me now.

closed 2022-04-10 by msinger