msinger/dmg-schematics #111: PPU OAM: Wrong /RD signals for OAM SRAMs

msinger, 2022-05-29:
On OAM sheet, I think the signal labeled OAM_CLK should be connected to the /OE pin of both OAM SRAMs instead of /OAM_A_RD and /OAM_B_RD. WUME (/OAM_B_RD) and WEMU (/OAM_A_RD) are not connected to the SRAMs. ZODO (OAM_CLK) is connected to them. I think those signals are also labeled wrong, because, an SRAM does not need any clock. I used OAM_CLK in my simulation for /OE and it fits perfectly. The signals /OAM_A_RD and /OAM_B_RD are just used to select which one of the two read bytes from the two SRAMs is actually driven onto the data bus D[0-7] during a read issued by the CPU.

rgalland, 2022-05-30:
This makes perfect sense. I am reorganising the sheets to lighten content on the OAM sheet especially. I also found a mistake with CARE that should be an AND_3 or not an OR_3. I actually had it written in my old notes.

msinger, 2022-06-02:
Why did you choose, the name /OAM_CS now, and not /OAM_RD or /OAM_OE? I find CS should only be used when the signal is active for both, read and write cycles.

rgalland, 2022-06-02:
I don't know to be honest. I should have named it /OAM_OE. It will be changed to that in my next commit

msinger, 2022-06-04:
I reopened so you don't forget the /OAM_OE :)

rgalland, 2022-06-05:
OK, changed for now. In the future, I think that adding the cells to the right of WAZU will show some sort of safety mechanism when writing at the same time as reading.

closed 2022-06-05 by rgalland