msinger/dmg-schematics #118: PPU: Reset signal names

msinger, 2022-06-04:
Maybe we could rename the reset signals RESET_VIDEO, /RESET_VIDEO, /RESET_VIDEO2, /RESET_VIDEO3, /RESET6, /RESET8 and /RESET9 to PPU_RSTn (n is 1-7). Because the numbering is messed up now. This would make it look similar to PPU_RD and PPU_WR, ...

Don't know what to do with RESET_VIDEO2, which comes from WINDOW DETECTION sheet. Maybe rename to WIN_RST or something. Or also move up to VIDEO CONTROL sheet.

rgalland, 2022-06-05:
Improvements made. I'll commit now so that you can have a look. I am still left with 2 signals named RESET6 and /RESET6 which could do with being named more wisely.

msinger, 2022-06-05:
Yes, much better! Maybe a good name for RESET6 would be PPU_HARD_RESET? Because it can't be triggered by software.

You didn't pick a new name for RESET_VIDEO (ATAR). Was this intentional? You don't need to, I'm just wondering...

closed 2022-07-02 by rgalland