msinger/dmg-schematics #129: SPRITE CTRL: OAM_PARSING signals

msinger, 2022-06-07:
On sprite control sheet, there are two signals, OAM_PARSING and /OAM_PARSING, that only differ in one having a bar on top of it. I find to a reader this indicates that those signals are always complementary towards each other, but this is not true. If I see it correctly, then /OAM_PARSING is two PPU clock cycles behind OAM_PARSING. I think they should have a more distinctive name then.

I have some proposals:
1. Since OAM_PARSING comes from a latch, we could call it OAM_PARSING_LATCHED.
2. Since /OAM_PARSING comes from a flip flop, we could call it /OAM_PARSING_BUFFERED, /OAM_PARSING_SYNC, /OAM_PARSING_REGISTERED, or some abbreviations of that.

I think only changing one name is sufficient.

rgalland, 2022-06-09:
@msinger , I've just pushed the changes. You may want to have a look before we close the ticket.

msinger, 2022-06-09:
Yes, nice find, that BUZA and AZEM are basically the same.

Why did you move it to LCD though? Was this the only space left where it fits?

rgalland, 2022-06-09:
Yes, it is mainly because there was still room on this sheet and quite a few common signals. This might still change by creating a new sheet but I'll look at it later. I'll close the issue for now.

closed 2022-06-09 by rgalland