msinger/dmg-schematics #146: IO pin names

msinger, 2022-06-10:
I would like to have the IO pins labeled the same as they are in the official Programming Manual on page 20: https://archive.org/details/GameBoyProgManVer1.1/page/n19/mode/2up

Right now, some are missing the bar on top (MCS, MWR, MOE). In the manual, MOE is labeled /MRD. This makes it consistent with the cartridge bus. On the PCB of the Game Boy Pocket, the test point on this trace is also labeled MRD. (http://iceboy.a-singer.de/db/mgb_iceboy_1/pcb_front.jpg)
Also in the Game Boy Color Service Manual, the pin names seem to match the ones from the Programming Manual: https://gamesx.com/wiki/lib/exe/fetch.php?media=schematics:gameboy_color_service_manual.pdf

You labeled /CS as /RAM_CS. I think it would make sense to add a comment next to it that says "RAM chip select" and then also next to A15 that says "ROM chip select".

As an exception, I think we should stick with PHI. I don't think it is a good idea to add Greek letters to KiCad.

I don't know how to best name the oscillator IO pins. We can leave it as it is or name it CK1_CK2, or just CK? I haven't figured out which one (CK1 or CK2) is output and which is input. In non-official documentation, they are mostly labeled XI and XO. I might have to measure it on a Game Boy Pocket. They are also labeled on its PCB.

I also think it may be good to have a comment next to each IO pin, that contains the pin number (or numbers in case of OSC).

When we start changing the SVG overlay later, we should not forget to name all the pins the same there.

Maybe we can add IO pads for GND and VDD also.

What do you think?

rgalland, 2022-06-11:
These are all good comments. I'll stat with renaming the pins according to the manual and adding notes as you suggested.

I can only imagine that CK1 is XO and CK2 is XO but I will need to check. In anycase, we can use CK1 and CK2 and add notes again.

Yes, it definitely make sense to have the GND and 5V pins to show that they do come from somewhere.

Let's leave this ticket open for now and perhaps use it to list all of the tasks to be done and then move to individual issues if they are big tasks. The overlay will be a big one.

msinger, 2022-06-11:
@Gekkio drew a pinout of the chip: https://github.com/Gekkio/gb-ctr/blob/master/images/DMG-CPU-pinout.pdf
According to this, pin 74 is CK1. I know from my experiments that pin 74 is the clock input and pin 73 is the output.
This means CK1 is XI and CK2 is XO. I don't know what you guessed in your comment, because you accidentally wrote XO twice. :grin:

What I also found while looking through his stuff, you can put nets with different names on one single bus now in KiCad, like he did here: https://github.com/Gekkio/gb-schematics/blob/main/MGB-xCPU/schematic/MGB-xCPU.pdf

rgalland, 2022-07-02:
I'm about o commit changes. Can you please check naming is suitable?

msinger, 2022-07-03:
Looks good

closed 2022-07-07 by rgalland