msinger/dmg-schematics #148: *_PHI signals

msinger, 2022-06-11:
Since the PHI IO pin itself is inverting the signal that goes through it, the net CPU_PHI1 should be called /CPU_PHI1 to be logically correct. And as a consequence, all other *_PHI signals that are derived from that need to be inverted too.

I think we should find another name for CPU_PHI2, because it looks different than the clock that comes out of the PHI pin. PHI is symmetric, CPU_PHI2 is not. I find if someone sees a signal called *_PHI in the schematics, they should be allowed to safely assume that it is related to what comes out of the PHI pin. One could argue, well, if the name ends with a number 2, then it is a different PHI, but this argument falls apart when looking at /DMA_PHI2, which is derived from DMA_PHI, which comes from /DMA_PHI, which comes from the symmetric CPU_PHI1.

rgalland, 2022-07-02:
This is a fair point. I'll invert the signals for now.

rgalland, 2022-07-02:
How about CPU_T4. This is quite a good name as we know that each CPU cycle takes 4 clock cycles. I'll commit now so you can have a look.

msinger, 2022-07-03:
Yes, I think this makes sense, because CPU_T4 is high during the 4th tick.

Still two things left here:
CPU_PHI1 can now be called just CPU_PHI, without the 1, because CPU_PHI2 is gone.
You forgot to remove the inversion from DMA_PHI and APU_PHI after inverting CPU_PHI1.

rgalland, 2022-07-08:
I fixed phi2 and at the same time reordered the sheets + export a pdf. Please let me know what you think.

msinger, 2022-07-08:
Yes, very good. I'm finally happy with this.

closed 2022-07-08 by rgalland