msinger/dmg-schematics #155: Various inverters with wrong type

msinger, 2022-09-09:
The following inverters have the wrong type/size:

RYCU inv_b -> inv_a
RAHU inv_b -> inv_c
SOHA inv_b -> inv_a
WADO inv_b -> inv_a
WEPO inv_b -> inv_a
WESA inv_b -> inv_a
WALO inv_b -> inv_a
XOLA inv_b -> inv_a
XERA inv_b -> inv_a
XENO inv_b -> inv_a
XUSY inv_b -> inv_a
XUPA inv_a -> inv_b
ABEF inv_a -> inv_b
AMEL inv_a -> inv_b
APUX inv_a -> inv_b
ATEJ inv_a -> inv_b
ASAH inv_a -> inv_b
AHAL inv_a -> inv_b
AZUB inv_a -> inv_b
ACAM inv_a -> inv_b
ADAZ inv_a -> inv_b
ZEME inv_c -> inv_d
ALER inv_d -> inv_c
ADAD inv_a -> inv_d
ASOP inv_a -> inv_b
BARU inv_b -> inv_a
BECY inv_a -> inv_c
COFU inv_b -> inv_a
DYLA inv_a -> inv_b
EVAJ inv_a -> inv_b
EJYB inv_a -> inv_c
FUTY inv_a -> inv_b
GEXY inv_b -> inv_a
HOVO inv_a -> inv_b

rgalland, 2022-09-10:
I'll let you check after the commit before closing the ticket.

msinger, 2022-09-10:
ATEJ is still A

rgalland, 2022-09-10:
Well done. I'll fix it later on.

Obtenir Outlook pour Android<https://aka.ms/AAb9ysg>

msinger, 2022-09-10:
I just found another one:
TONA  B -> A

rgalland, 2022-09-11:
I should be fine after the commit.

closed 2022-09-12 by rgalland