msinger/dmg-schematics #161: MODE2 wire is mislabeled as MODE1

tclemens, 2023-05-25:
It's source is `OAM_PARSING && !DMA_RUN`. It's also a component of `STAT[1]` which should only be set in modes 2 and 3.

msinger, 2023-05-25:
Thank you, I think you are right. Seems like we have overlooked this. If it raises bit 1 of the MODE[1:0] field in the STAT register, then it must be high during MODE2, not MODE1. I will check simulation results once more just to make sure, and then fix it.

msinger, 2023-05-25:
Hi @rgalland, thanks, I'm already done. Just about to push. :)

closed 2023-05-25 by msinger

rgalland, 2023-05-25:
@msinger , man! you were fast.

msinger, 2023-05-25:
Let's hope I didn't miss something.

msinger, 2023-05-25:
Changing INT_VBL_BUF to INT_VBL would clean up FF0F INT (page 10). This was bothering me for a while that this is the only interrupt that hat a _BUF in its name.

msinger, 2023-05-25:
Oh, careful. :)
You mean, INT_HBL to MODE0.

rgalland, 2023-05-25:
Yes, that's what I meant.

commit cd74ea66 (2023-05-25): Rename wire MODE1 -> MODE2

Wire was mislabeled. Found by @tclemens.

fixes #161