msinger/dmg-schematics #162: MOTY

paulb-nl, 2023-10-22:
The MOTY signal doesn't make much sense to me.

When MOTY is low it will drive the external data bus and when it is high it will enable pullups. 

However when it is high it will also drive /RD low.  The problem is that MOTY will go high even if only NON_VRAM_MREQ is low because of MOCA.

Because of that LAGU also doesn't make sense. (READ & ~NON_VRAM_MREQ). If this is high then MOTY should go low but this is impossible.

It seems to me /RD should only go low with something like (MOTY & READ & NON_VRAM_MREQ);

msinger, 2023-10-22:
It's okay that /RD is low, even if there is no actual read access to the cartridge or WRAM. If you measure at the real Game Boy, you will see that /RD is always low, except if there is a write access to the cartridge or WRAM going on.

Even between two consecutive writes, the /RD signal is shortly low on the real hardware, like low is the default state of /RD if the cartridge bus is idle. This is how it looks like in the simulation:
[image]
(TEXO == NON_VRAM_MREQ)

The /RD signal is glitchy, it has short 1-hazards. We see such spikes on the real hardware as well, but if I remember correctly, at a different position, maybe due to inaccurate timing parameters.

I think you can see /RD just as an inverted state of /WR, but with different transition times. /RD goes high roughly 3/8 phi periods before /WR goes low. And /RD goes back low, roughly 1/8 phi periods after /WR went high.

> It seems to me /RD should only go low with something like (MOTY & READ & NON_VRAM_MREQ);

I agree with that, but for some reason Sharp decided that /RD should be low, even if there is no read operation going on. I don't know why they did this.

Please let me know if I misunderstood something.

paulb-nl, 2023-10-22:
Well that was unexpected. Thanks for the information.

I had noticed that Gekkio's hdl model of the SM83 CPU outputs READ only when PHI is low so maybe they did it like this because they needed an earlier/longer /RD signal.

closed 2023-10-23 by paulb-nl

msinger, 2023-10-23:
Yes, that could be a reason.