msinger/dmg-schematics #163: LYC int RUPO PAGO

paulb-nl, 2023-11-19:
FF41 bit 2 (RUPO) is set by ROPO (LYC int). That makes sense but then at the same time it is reset by PAGO if `nLYC_STAT_RESET` is high or not writing to FF41.

So the only way RUPO is not reset is during a PPU hard reset and also writing to FF41 which of course is impossible.

Is it actually being reset all the time? What does that NOR latch output when it is reset and set at the same time?

As we know from the Mooneye test suite, FF41 bit 2 is slightly delayed with a 0->1 transition compared to 1->0 where it reads a NOP later during the lyc0->1 test. Is this related?
https://github.com/Gekkio/mooneye-test-suite/blob/8d742b9d55055f6878a2f3017e0ccf2234cd692c/acceptance/ppu/lcdon_timing-GS.s#L108

msinger, 2023-11-19:
If both inputs of RUPO (FF41 bit 2) are 1 at the same time, both NOR gates that make up the latch should output a zero. So Q and /Q should be 0 at the same time.

I see in the simulation that RUPO.Q is always 0 and RUPO.~Q is the inverted state of ROPO (LYC int). So after the tri-state inverter SEGO, the value read from FF41 bit 2 should always be just the same as ROPO. (I haven't tested the case where nLYC_STAT_RESET and RYJU are both 0 though.)

I don't know why they put the latch here. Should also work without it. Maybe it has "historic" reasons like many of such nonsensical looking things.

I can't tell where the slight 0->1 delay compared to 1->0 comes from. That's too complicated for me now. :)

msinger, 2024-06-26:
I close the issue, since I don't think that there is anything to fix. If I missed something or if you have further questions, feel free to reopen it.

closed 2024-06-26 by msinger