msinger/dmg-schematics #164: Fuse repos

ogamespec, 2024-06-24:
Suggestion to merge all DMG research repositories into one common repository. 
From the current list of repositories it is not very clear what each repository is for, as they are all used in some way for DMG research and DMG circuit reconstruction.

msinger, 2024-06-26:
Hm, I was thinking for a while now. I wouldn't want to throw everything together. I still have nightmares from 2018 when I was forced to use CVS at work. One huge garbage shit pile repository.

It would make sense to move the cell documentation to the schematics repo, which already contains the cell layouts.
And also to move the Wide-Boy doc to the wideboy repo. But I wouldn't want to merge the dmg schematics with the Wide-Boy stuff for example. The gbreveng repo is kind of a dumpster of experiments and it also contains html docs. I think the dmg schematics repo should only contain the final product (schematics, layout, docs) for people that are just interested in that.

I can also try to add better README.md files to all repos and also give them a better short description. I don't know what a github project would offer, if that helps in organizing the repos. What do you think?

Can you tell me what the actual thing was that may have frustrated you in your search? Was it to find the repo that contains the html files of the cell documentation for creating the issue with the missing piece of metal?

ogamespec, 2024-06-26:
In general, I agree, WideBoy can be made separately and not mixed in a heap.

> Can you tell me what the actual thing was that may have frustrated you in your search?
Well, here's the easy part. The first thing I would like to know is where is the PPU, where is the APU (I mean standard cell domains).
I would also like to have the pinout of the chip at hand.
Also on the `map` I would like to see the name of the interconnects so I can match the pdf schematic with what I see on the `map`. 
The cell description is on the website for some reason, but the pdf schematic is on GitHub. The `map` is also on the site and there is no way to look at the topology without a browser.

I guess it's subjective :)

ogamespec, 2024-06-26:
It's also not very clear where the "landing page" is. Where is the entrance to the "portal" so to speak? :)

ogamespec, 2024-06-27:
I think I found the landing page:

https://github.com/msinger/dmg-schematics/blob/master/README.md

msinger, 2024-06-27:
The landing page is supposed to be the [index page](http://iceboy.a-singer.de) of my website. But it was originally only about my FPGA implementation. Now everything else was added there. I'm not good at this. Yes, a good documentation would have an overview of the chip first maybe.

I don't know how I can do the net names on the map yet, but I'd like to have that too.

I thougt having a browser map would make it more accessible to everyone, because it just works. Before that we had the overlay SVG from Furrtek that you needed to align with the die shot in Inkscape. I did't like that, it was laggy all the time. I like the approach of having the actual information in text form from where it can be transformed. If you want to look at the huge PNG files locally without a browser, you can do that. You can use the [nlconv](http://github.com/msinger/nlconv) tool to convert the netlists to PNGs and then open them in Gimp or something similar that can handle huge images (16k x 16k pixels). I have actually documented the tool in the README.md file of that repository. If you can't get the Mono/C# stuff running, I can upload the images for you somewhere.

Yes, but the README.md in the dmg-schematics repo just describes the contents of that repo. Not the others. I don't think there is currently an overview for everything. It's so much to do. I always think I will make everything look good when I'm done with the real work. :)

ogamespec, 2024-06-27:
> If you want to look at the huge PNG files locally without a browser, you can do that.

Would it be possible to add something like this to the Release section on GitHub? It would be more convenient for me to study exactly the picture, right from my phone, instead of loading the site every time.
And if possible without a chip photo, there's no use for it. You can just solid background or even transparent, then PNG will not be big size.

msinger, 2024-06-27:
There is no chip photo in the background. The tool only generates the transparent overlays. I assume one PNG containing all four layers is convenient for you? Or do you want one PNG for each layer separately?

ogamespec, 2024-06-27:
It's better all in one, it's more convenient.

msinger, 2024-06-27:
I added the PNG to the [latest release](https://github.com/msinger/dmg-schematics/releases/tag/v0.6).

ogamespec, 2024-06-27:
> I added the PNG to the [latest release](https://github.com/msinger/dmg-schematics/releases/tag/v0.6).

Cool, thanks!