msinger/dmg-schematics #165: Page 25 issues

ogamespec, 2025-03-23:
Hi, I found some inaccuracies on Page 26 and also added the signal names there. Please check :)

[image]

My annotated schematic can be found here:

https://github.com/emu-russia/dmgcpu/blob/main/wiki/soc/arb.md

msinger, 2025-03-23:
So you 'fixed' those two things:
 * There is a connection from `SALE`.`Q` to `SERE`.`IN2`
 * The TRI_INVs on the right hand side that drive the D[0..7] and MD[0..7] have the types:
    - INV_IF0, if they are in column R.
    - INV_IF1, if they are in column T.

Did I miss something?

So, I'm confused, because I see both things differently. I use the following trick to determine the polarity of the enable input of the TRI_INVs:

[image]

I took a look at `RAKU` and `TOKU`. Both of them have the dots on the same side, so they are IF1 in my opinion. Do you use a different approach in identifying those cells? Like I already listed it above, I noticed that you marked all TRI_INVs starting with an R as IF0, so maybe you switched GND and VCC in your mind when identifying those cells?

If you think about it, it makes no sense to have them enable on different logic levels. The ones that are driving D[0..7] all have the same signal on their enable input. If only some of the enable inputs were inverted, then there would be something driven onto D[0..7] continuously, and no other part of the SoC would be able to ever drive a whole byte onto D[0..7].

The `SALE`-`SERE` connection I also can't see:

[image]

I think there is a corner in the encircled area that connects from south to east. No straight connection from west to east.

Your signal names all make sense to me, btw. 😄

ogamespec, 2025-03-23:
SALE-SERE situation is actually my fault, sorry:

[image]

I'll double-check the tristates and write it up.

ogamespec, 2025-03-23:
on tristates. I distinguish them by the m1, which is shaped like a pistol:

[image]

My RAKU and TOKU (DMG CPU RevA):

[image]
(as you can see RAKU = notif0, TOKU = notif1)

it seems to be the revision of the CPU...

msinger, 2025-03-23:
I understand what you mean. But I don't know yet if it is true or false. Your pistol contains big and small vias. Sometimes the big via is at the tip, other times the small via is at the tip. This would mean that there are four different types of those TRI_INV cells, not just two. Because we would have:
 * Pistol to GND with big tip.
 * Pistol to GND with small tip.
 * Pistol from GND with big tip.
 * Pistol from GND with small tip.

I will search for the raw image you used for tracing those cells. Maybe you made a mistake there, or maybe we discovered new cell types. But as I argued before, those instances in this circuit here can't have opposite polarity, otherwise the data bus would be permanently blocked.

msinger, 2025-03-23:
Are all pictures you made from a rev A? Or did you open two different chips?

msinger, 2025-03-23:
I took pictures of RAKU, the left one is from your Cells_5_Fused.jpg that you took from the active/poly layer:

[image]

It points in the opposite direction than you have drawn it in your overlay layout. In Cells_5_Fused.jpg you can clearly see the three small vias. The big via is between metal, so it can't be seen anymore.

msinger, 2025-03-24:
On second glance, I think I interpreted the pistols different than you (made up by different metal parts), but it is still the opposite direction. 😆

ogamespec, 2025-03-24:
> Because we would have: ..

Can you show examples? (and have the gnd on the left).

> Are all pictures you made from a rev A? Or did you open two different chips?

I reconstruct the cell topology (poly) using RevB. And over RevA I mark the cells.
I use: https://siliconpr0n.org/map/nintendo/dmg-cpu-a/

I think you have the gnd on the right side in one place in the image, that's why it's confusing:

[image]

msinger, 2025-03-24:
No, I don't have examples for the four types. It was just an idea, trying to explain how they could still have the same polarity. There are most likely just two types. Since I understood your pistol in the wrong way, it wouldn't have been the tip of the pistol that was big or small, instead it would have been the trigger that was flipped around: Big hole at the front vs. small hole at the front. (Sorry, I don't know much about weapons, maybe my terminology is wrong. By 'trigger' I mean where you put your index finger through to fire the pistol.)

Considering your provided depiction of the pistol and looking at the rev A image that you used, I believe I still see the RAKU pistol point in the opposite direction that you:

[image]

ogamespec, 2025-03-24:
[image]

I double-checked, and it's my fault. I apologize, I think we can close Issue, but I'll be back :)

closed 2025-03-24 by msinger